ros-industrial/industrial_ci #852: ImportError: cannot import name 'missing_whitespace_around_operator' from 'pycodestyle'

cardboardcode, 2024-03-16:
## **Issue Description** :bug: :page_facing_up: 

Attempting to run `industrial_ci` using **ROS 2 Rolling** version causes Continuous Integration (CI) to fail the  due to the following error:

```bash
ImportError: cannot import name 'missing_whitespace_around_operator' from 'pycodestyle' (/usr/lib/python3/dist-packages/pycodestyle.py). Did you mean: 'whitespace_around_operator'?
```

```bash
flake8.exceptions.FailedToLoadPlugin: Flake8 failed to load plugin "pycodestyle" due to cannot import name 'missing_whitespace_around_operator' from 'pycodestyle' (/usr/lib/python3/dist-packages/pycodestyle.py).
```

<details>
  <summary>Click here for <b>Raw CI Report</b> [Cropped]</summary>

```bash
  5: Traceback (most recent call last):
  5:   File "/usr/lib/python3/dist-packages/flake8/plugins/finder.py", line 296, in _load_plugin
  5:     obj = plugin.entry_point.load()
  5:           ^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
  5:     module = import_module(match.group('module'))
  5:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
  5:     return _bootstrap._gcd_import(name[level:], package, level)
  5:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  5:   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  5:   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  5:   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  5:   File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  5:   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  5:   File "/usr/lib/python3/dist-packages/flake8/plugins/pycodestyle.py", line 25, in <module>
  5:     from pycodestyle import missing_whitespace_around_operator as _missing_whitespace_around_operator  # noqa: E501
  5:     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5: ImportError: cannot import name 'missing_whitespace_around_operator' from 'pycodestyle' (/usr/lib/python3/dist-packages/pycodestyle.py). Did you mean: 'whitespace_around_operator'?
  5: 
  5: During handling of the above exception, another exception occurred:
  5: 
  5: Traceback (most recent call last):
  5:   File "/opt/ros/rolling/bin/ament_flake8", line 33, in <module>
  5:     sys.exit(load_entry_point('ament-flake8==0.16.3', 'console_scripts', 'ament_flake8')())
  5:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/opt/ros/rolling/lib/python3.12/site-packages/ament_flake8/main.py", line 31, in main
  5:     rc, _ = main_with_errors(argv=argv)
  5:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/opt/ros/rolling/lib/python3.12/site-packages/ament_flake8/main.py", line 88, in main_with_errors
  5:     report = generate_flake8_report(
  5:              ^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/opt/ros/rolling/lib/python3.12/site-packages/ament_flake8/main.py", line 267, in generate_flake8_report
  5:     style = get_flake8_style_guide(flake8_argv)
  5:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/opt/ros/rolling/lib/python3.12/site-packages/ament_flake8/main.py", line 166, in get_flake8_style_guide
  5:     application.find_plugins(
  5:   File "/usr/lib/python3/dist-packages/flake8/main/application.py", line 128, in find_plugins
  5:     self.plugins = finder.load_plugins(raw, opts)
  5:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/usr/lib/python3/dist-packages/flake8/plugins/finder.py", line 370, in load_plugins
  5:     return _classify_plugins(_import_plugins(plugins, opts), opts)
  5:                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  5:   File "/usr/lib/python3/dist-packages/flake8/plugins/finder.py", line 312, in _import_plugins
  5:     return [_load_plugin(p) for p in plugins]
  5:             ^^^^^^^^^^^^^^^
  5:   File "/usr/lib/python3/dist-packages/flake8/plugins/finder.py", line 298, in _load_plugin
  5:     raise FailedToLoadPlugin(plugin.package, e)
  5: flake8.exceptions.FailedToLoadPlugin: Flake8 failed to load plugin "pycodestyle" due to cannot import name 'missing_whitespace_around_operator' from 'pycodestyle' (/usr/lib/python3/dist-packages/pycodestyle.py).
``` 
</details>

Reference Link to **Raw CI Report**: [Click here](https://github.com/cardboardcode/virtual_camera/actions/runs/8275558270/job/22659358345) for full details.

The `industrial_ci_action.yml` used is as follows:

```bash
name: CI

on: [push, pull_request]

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: rolling, ROS_REPO: testing}
          - {ROS_DISTRO: rolling, ROS_REPO: main}
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v3
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{matrix.env}}
```
**Reference Link**: https://github.com/cardboardcode/virtual_camera/blob/rolling_devel/.github/workflows/industrial_ci_action.yml

## **Expected Behaviour** :green_circle: 

CI passes with **flake8** test passing without error.

## **Actual Behaviour** :red_circle: 

CI fails even though, for the same code on other versions of ROS 2 used for **Humble**, `flake8` is passing successfully. This can be implied that the root of the issue is due to a potential discrepancy in **pycodestyle** version used for **industrial_ci**'s ROS 2 Rolling Ridley.

## **Deduction of Potential Root Cause** :mag: 
Investigating the issue further (by googling), it seems that the issue of **cannot import name 'missing_whitespace_around_operator' from 'pycodestyle'** stems from the `pycodestyle` version, according to [this thread](https://bbs.archlinux.org/viewtopic.php?id=287648).

## **Remarks** :speech_balloon: 
Wondering if anyone is also encountering this issue using **ROS 2 Rolling Ridley** in `industrial_ci`.

The solution to this issue may lie in installing the specific compatible `pycodestyle` module version when running CI in **ROS 2 Rolling Ridley** in `industrial_ci_action.yml` which I think is better-handled on the backend here by maintainers since it should be affecting everyone who uses the same configuration.

Will delve into the inner workings of `industrial_ci` here when I am free to see if I can tweak any backend settings and make a pull request to change it for all users if warranted.

mathias-luedtke, 2024-03-19:
Hi, based on your description, I doubt that this is an issue in industrial_ci.
I think it might be related to https://bugs.launchpad.net/ubuntu/+source/pycodestyle/+bug/2053217

cardboardcode, 2024-03-19:
Noted. Closing the issue due to invalidity.

closed 2024-03-19 by cardboardcode