ros-industrial/industrial_ci #855: Random failure when installing dependencies

bijoua29, 2024-03-27:
Since yesterday, we are getting random failures when installing dependencies either upstream or target dependencies. Here's an example for installl_upstream_dependencies:

  executing command [apt-get install -y -qq ros-rolling-rclcpp]
  E: Unable to locate package ros-rolling-rclcpp
  ERROR: the following rosdeps failed to install
    apt: command [apt-get install -y -qq ros-rolling-rclcpp] failed
  '( source /opt/ros/rolling/setup.bash && rosdep install -q --from-paths /home/runner/work/schilling_rovs/schilling_rovs/.work/upstream_ws/src --ignore-src -y | grep -E '(executing command)|(Setting up)' ; )' returned with 1
'install_upstream_dependencies' returned with code '1' after 0 min 9 sec

The particular package that it fails to fins is random and different on every run. All these packages are available locally at my desktop when I query.

Any idea what is going on as none of our CI builds are working since yesterday.

mathias-luedtke, 2024-03-27:
rolling was broken for some time, so I configured the building repository as the apt source.
However, the building repo might be quite unstable, so it is not ideal.
I have reverted it now (https://github.com/ros-industrial/industrial_ci/pull/851).
Hopefully this will fix you builds.

bijoua29, 2024-03-28:
@mathias-luedtke Yes, Thanks. This fixed our builds.

bijoua29, 2024-03-28:
@mathias-luedtke Spoke too soon. Still getting errors in our build:

  executing command [apt-get install -y -qq ros-rolling-rqt-gui]
  E: Unable to correct problems, you have held broken packages.
  ERROR: the following rosdeps failed to install
    apt: command [apt-get install -y -qq ros-rolling-rqt-gui] failed
  '( source /opt/ros/rolling/setup.bash && rosdep install -q --from-paths /home/runner/work/xxx/xxx/.work/upstream_ws/src --ignore-src -y | grep -E '(executing command)|(Setting up)' ; )' returned with 1
'install_upstream_dependencies' returned with code '1' after 1 min 35 sec

danthony06, 2024-03-29:
Is it related to this? https://discourse.ros.org/t/psa-rolling-ci-or-docker-build-fix-from-rosdep-errors-in-24-04-transition/36902

mathias-luedtke, 2024-03-29:
No, this is for running rolling on `jammy`. 
I tried that weeks ago in https://github.com/ros-industrial/industrial_ci/pull/850 and it did not work..

mathias-luedtke, 2024-03-29:
#858 adds support for `ROS_DISTRO=rolling OS_CODE_NAME=jammy`
There is not much we can do about `noble` because it is still being migrated

bijoua29, 2024-04-01:
The failures now seem not just restricted to rolling packages. For example, I am getting the following during the 'init' step:

```
$ sudo apt-get -qq install -y --no-upgrade --no-install-recommends apt-utils build-essential gnupg2 dirmngr ca-certificates | grep -E 'Setting up' 
  E: Unable to correct problems, you have held broken packages.
  'sudo apt-get -qq install -y --no-upgrade --no-install-recommends apt-utils build-essential gnupg2 dirmngr ca-certificates | grep -E 'Setting up' ' returned with 100
'init' returned with code '100' after 0 min 1 sec
```

mathias-luedtke, 2024-04-01:
> The failures now seem not just restricted to rolling packages. For example, I am getting the following during the 'init' step:

Yes, my pipelines are failing as well.
But `noble` has not even been released yet so it is expected to be unstable.

If you want something stable, set `OS_CODE_NAME=jammy`

bijoua29, 2024-04-01:
You are right @mathias-luedtke . Can't trust noble right now.

I need to be able to use the rolling sync before the noble update. I'm hoping the process described in https://discourse.ros.org/t/psa-rolling-ci-or-docker-build-fix-from-rosdep-errors-in-24-04-transition/36902 works

bijoua29, 2024-04-05:
Since this is now a Ubuntu Noble stability problem before Apr 25,2024 release, we can close out this issue.

bijoua29, 2024-04-05:
Closing now

closed 2024-04-05 by bijoua29