ros-industrial/industrial_ci #866: Python setuptools package conflict with PRERELEASE

johnwason, 2024-08-01:
I am seeing a strange conflict with the Python `setuptools` package when `PRERELEASE` is set to true. I re-ran the test on an old branch that was working in the past, and am seeing the error where it did not occur before.

- Failing run: https://github.com/johnwason/robotraconteur/actions/runs/10193062593/job/28197014982
- Old passing run: https://github.com/robotraconteur/robotraconteur/actions/runs/8955115165/job/24595493073
- Branch: https://github.com/robotraconteur/robotraconteur/tree/ros2-iron
- Workflow: https://github.com/robotraconteur/robotraconteur/blob/ros2-iron/.github/workflows/ros_main.yaml

It looks like it is finding a version of setuptools in `/usr/local/lib/python3.10`. This is probably conflicting with the distribution version of setuptools installed using `apt install python3-setuptools`.

Here is the relevant log section:

```
2024-08-01T06:27:01.8118366Z -- Running setup.py in /tmp/ws/build_isolated/robotraconteur/out/Python3
2024-08-01T06:27:01.8119379Z running install
2024-08-01T06:27:01.8120724Z /usr/local/lib/python3.10/dist-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
2024-08-01T06:27:01.8121926Z !!
2024-08-01T06:27:01.8122111Z 
2024-08-01T06:27:01.8122329Z         ********************************************************************************
2024-08-01T06:27:01.8122926Z         Please avoid running ``setup.py`` directly.
2024-08-01T06:27:01.8123524Z         Instead, use pypa/build, pypa/installer or other
2024-08-01T06:27:01.8124100Z         standards-based tools.
2024-08-01T06:27:01.8124362Z 
2024-08-01T06:27:01.8124897Z         See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
2024-08-01T06:27:01.8125689Z         ********************************************************************************
2024-08-01T06:27:01.8126151Z 
2024-08-01T06:27:01.8126263Z !!
2024-08-01T06:27:01.8126591Z   self.initialize_options()
2024-08-01T06:27:01.8127070Z Traceback (most recent call last):
2024-08-01T06:27:01.8127926Z   File "/tmp/ws/build_isolated/robotraconteur/out/Python3/setup.py", line 17, in <module>
2024-08-01T06:27:01.8128920Z     setup(name='RobotRaconteur',
2024-08-01T06:27:01.8129850Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/__init__.py", line 108, in setup
2024-08-01T06:27:01.8130718Z     return distutils.core.setup(**attrs)
2024-08-01T06:27:01.8131735Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 184, in setup
2024-08-01T06:27:01.8132712Z     return run_commands(dist)
2024-08-01T06:27:01.8133800Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 200, in run_commands
2024-08-01T06:27:01.8134828Z     dist.run_commands()
2024-08-01T06:27:01.8135890Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 970, in run_commands
2024-08-01T06:27:01.8136940Z     self.run_command(cmd)
2024-08-01T06:27:01.8137931Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/dist.py", line 945, in run_command
2024-08-01T06:27:01.8138898Z     super().run_command(command)
2024-08-01T06:27:01.8140020Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 988, in run_command
2024-08-01T06:27:01.8141066Z     cmd_obj.ensure_finalized()
2024-08-01T06:27:01.8142224Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
2024-08-01T06:27:01.8143291Z     self.finalize_options()
2024-08-01T06:27:01.8144460Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/command/install.py", line 62, in finalize_options
2024-08-01T06:27:01.8145561Z     super().finalize_options()
2024-08-01T06:27:01.8146804Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/command/install.py", line 407, in finalize_options
2024-08-01T06:27:01.8148099Z     'dist_fullname': self.distribution.get_fullname(),
2024-08-01T06:27:01.8149365Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 266, in get_fullname
2024-08-01T06:27:01.8150570Z     return _distribution_fullname(self.get_name(), self.get_version())
2024-08-01T06:27:01.8151932Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 284, in _distribution_fullname
2024-08-01T06:27:01.8153164Z     canonicalize_version(version, strip_trailing_zero=False),
2024-08-01T06:27:01.8154778Z TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
2024-08-01T06:27:01.8155757Z CMake Error at RobotRaconteurPython3/python3_install.cmake:16 (message):
2024-08-01T06:27:01.8156388Z   Python install failed
2024-08-01T06:27:01.8156755Z Call Stack (most recent call first):
2024-08-01T06:27:01.8157273Z   RobotRaconteurPython3/cmake_install.cmake:46 (include)
2024-08-01T06:27:01.8158086Z   cmake_install.cmake:65 (include)
2024-08-01T06:27:01.8158338Z 
2024-08-01T06:27:01.8158342Z 
2024-08-01T06:27:01.8158490Z ---
2024-08-01T06:27:01.8158897Z --- stderr: robotraconteur
2024-08-01T06:27:01.8159247Z RobotRaconteur Python Package Version 1.2.0
2024-08-01T06:27:01.8159624Z CMake Warning:
2024-08-01T06:27:01.8160051Z   Manually-specified variables were not used by the project:
2024-08-01T06:27:01.8160409Z 
2024-08-01T06:27:01.8160510Z     CATKIN_SKIP_TESTING
2024-08-01T06:27:01.8160696Z 
2024-08-01T06:27:01.8160700Z 
2024-08-01T06:27:01.8161668Z /tmp/ws/src/robotraconteur/generated_src/Python3/swigwordsize64/RobotRaconteurPythonPYTHON_wrap.cxx: In function ‘PyObject* PyInit__RobotRaconteurPython()’:
[... 31 lines cut ...]
2024-08-01T06:27:01.8180745Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 200, in run_commands
2024-08-01T06:27:02.9250191Z     dist.run_commands()
2024-08-01T06:27:02.9251812Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 970, in run_commands
2024-08-01T06:27:02.9252910Z     self.run_command(cmd)
2024-08-01T06:27:02.9253850Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/dist.py", line 945, in run_command
2024-08-01T06:27:02.9254785Z     super().run_command(command)
2024-08-01T06:27:02.9255670Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 988, in run_command
2024-08-01T06:27:02.9256283Z     cmd_obj.ensure_finalized()
2024-08-01T06:27:02.9256949Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
2024-08-01T06:27:02.9257560Z     self.finalize_options()
2024-08-01T06:27:02.9258276Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/command/install.py", line 62, in finalize_options
2024-08-01T06:27:02.9258917Z     super().finalize_options()
2024-08-01T06:27:02.9259637Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/command/install.py", line 407, in finalize_options
2024-08-01T06:27:02.9260400Z     'dist_fullname': self.distribution.get_fullname(),
2024-08-01T06:27:02.9261126Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 266, in get_fullname
2024-08-01T06:27:02.9261849Z     return _distribution_fullname(self.get_name(), self.get_version())
2024-08-01T06:27:02.9262688Z   File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 284, in _distribution_fullname
2024-08-01T06:27:02.9263406Z     canonicalize_version(version, strip_trailing_zero=False),
2024-08-01T06:27:02.9264076Z TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
2024-08-01T06:27:02.9264765Z CMake Error at RobotRaconteurPython3/python3_install.cmake:16 (message):
2024-08-01T06:27:02.9265226Z   Python install failed
2024-08-01T06:27:02.9265509Z Call Stack (most recent call first):
2024-08-01T06:27:02.9265908Z   RobotRaconteurPython3/cmake_install.cmake:46 (include)
2024-08-01T06:27:02.9266304Z   cmake_install.cmake:65 (include)
```

mathias-luedtke, 2024-08-03:
Thanks for reporting!
It looks like the Github images/VM now bundles its own version of  `setuptools`.

mathias-luedtke, 2024-08-04:
> It looks like the Github images/VM now bundles its own version of setuptools.

no, it is more complicated..
The latest release of ros_buildfarm is more than 5 years old (https://github.com/ros-infrastructure/ros_buildfarm/issues/963) and it upgrades setuptools during the build process.
And on July 29th https://pypi.org/project/setuptools/72.1.0/ got released, which breaks the ROS2 builds.

https://github.com/ros-industrial/industrial_ci/pull/870 now uses the latest development versions of ros_buildfarm, which should fix the setuptools issue.

@johnwason: Please run your jobs again and close this issue if they pass now.

johnwason, 2024-08-05:
The prerelease tests are working now. Thanks!

closed 2024-08-05 by johnwason