ros-industrial/industrial_ci #867: Prerelease error with Jazzy

johnwason, 2024-08-01:
I am seeing an error when using `PRERELEASE` set to true with Jazzy. The other distributions are working normally except for the problem described in #866. The non-prerelease is working normally.

Full log: https://github.com/johnwason/robotraconteur/actions/runs/10192639718/job/28195853746

Relevant log section:

```
Step 9/26 : RUN useradd -u 1000 -m buildfarm
   ---> Running in bc3de201b03a
  useradd: UID 1000 is not unique
   ---> Removed intermediate container bc3de201b03a
  The command '/bin/sh -c useradd -u 1000 -m buildfarm' returned a non-zero code: 4
```

mathias-luedtke, 2024-08-03:
Thanks for reporting!
If if I am not mistaken, this is actually an issue in https://github.com/ros-infrastructure/ros_buildfarm/
I will have a closer look!

mathias-luedtke, 2024-08-03:
See https://github.com/ros-infrastructure/ros_buildfarm/issues/1060 as well.

mathias-luedtke, 2024-08-04:
@johnwason: I deployed a work-around, please confirm that this fixes your build

johnwason, 2024-08-05:
The prerelease tests are working now. Thanks!

closed 2024-08-05 by johnwason