ros-industrial/industrial_ci #876: Multiple packages found with the same name

dave992, 2024-08-15:
I run in the following error during the `install_target_dependencies` step when running the CI:
```bash
$ ( source /builds/ros_meta_pkg/workspaces/upstream_ws/install/setup.bash && rosdep install --from-paths /builds/ros_meta_pkg/workspaces/target_ws --ignore-src -y | grep -E '(executing command)|(Setting up)' ; )
ERROR: Rosdep experienced an error: Multiple packages found with the same name "ros_pkg_1":
- ros_pkg_1
- workspaces/target_ws/src/ros_meta_pkg/ros_pkg_1
Multiple packages found with the same name "ros_pkg_2":
- ros_pkg_2
- workspaces/target_ws/src/ros_meta_pkg/ros_pkg_2
```

## CI Configuration
The GitLab CI settings I use:
```yaml
# Source: https://github.com/ros-industrial/industrial_ci/issues/627
ros_industrial_ci:
  image: ros:humble
  before_script:
    - apt-get -qq update && apt-get -qq install bash coreutils git grep # install industrial_ci dependencies
    - git config --add --global "url.${CI_SERVER_PROTOCOL}://gitlab-ci-token:${CI_JOB_TOKEN}@${CI_SERVER_HOST}:${CI_SERVER_PORT}/.insteadOf" "${CI_SERVER_URL}"
    - git config --add --global "url.${CI_SERVER_PROTOCOL}://gitlab-ci-token:${CI_JOB_TOKEN}@${CI_SERVER_HOST}:${CI_SERVER_PORT}/.insteadOf" "ssh://git@${CI_SERVER_HOST}:"
    - git config --add --global "url.${CI_SERVER_PROTOCOL}://gitlab-ci-token:${CI_JOB_TOKEN}@${CI_SERVER_HOST}:${CI_SERVER_PORT}/.insteadOf" "ssh://git@${CI_SERVER_HOST}/"
    - git clone --quiet --depth 1 https://github.com/sam-xl/industrial_ci .industrial_ci -b fix/rosdep_multiple_pkgs_found
  script: .industrial_ci/gitlab.sh
  variables:
    BASEDIR: "${CI_PROJECT_DIR}/workspaces"
    CCACHE_DIR: "${CI_PROJECT_DIR}/ccache"
    ISOLATION: "shell"
  cache:
    # https://docs.gitlab.com/ee/ci/caching/#sharing-caches-across-different-branches
    key: "${CI_JOB_NAME}"
    paths:
      - ccache
```

I took this config from #627 

## Observations
I only run into this error when I set `$BASEDIR` inside of `$CI_PROJECT_DIR`. Using `$BASEDIR: /workspaces` does not result in this error.

Interestingly enough, if I manually log into the container created by the CI job (by keeping it alive after the failure) and run the same command and from the same directory, I do not get any errors. 

The command:

    $ ( source /builds/ros_meta_pkg/workspaces/upstream_ws/install/setup.bash && rosdep install --from-paths /builds/ros_meta_pkg/workspaces/target_ws --ignore-src -y | grep -E '(executing command)|(Setting up)' ; )

The only way to replicate the same error is to run:

    $ rosdep install --from-paths /builds/ros_meta_pkg --ignore-src -y

or 

    $ rosdep install --from-paths . --ignore-src -y

I do not understand why this behaves like this. Even printing the `$ROS_PACKAGE_PATH` shows no unexpected folders.

### Folder structure
The folder structure looks like this: 
```yaml
ros_meta_pkg (repository)
- ros_pkg_1
- ros_pkg_2
- workspaces (created by industrial ci)
  - upstream_ws  
  - target_ws
```
Commands get executed from the `ros_meta_pkg` folder.

## Possible solution
If I run the rosdep command from the `target_ws` I no longer run into any issues. 

Looking at the source code this can be achieved by removing the `/src` from the following line. This step can probably be skipped. 
https://github.com/ros-industrial/industrial_ci/blob/d23b9ad2c63bfad638a2b1fe3df34b8df9a2f17b/industrial_ci/src/workspace.sh#L396

And replacing `ici_exec_in_workspace "$extend" "."` with `ici_exec_in_workspace "$extend" "$@"` so the command switches to the `target_ws` folder before executing the `rosdep` command
https://github.com/ros-industrial/industrial_ci/blob/d23b9ad2c63bfad638a2b1fe3df34b8df9a2f17b/industrial_ci/src/workspace.sh#L376

mathias-luedtke, 2024-12-21:
Thanks for reporting!

Unfortunately, I have not found any proper solution for this issue yet.
The recommended work-around is to set
https://github.com/ros-industrial/industrial_ci/blob/53cd3f10a847e3d374ce385f447587ebcb6791d7/.gitlab-ci.yml#L81

This is not be documented properly, though.

ErichFoster-Havoc, 2025-08-21:
I have a similar issue and I tried to use `BASEDIR` in a gh action, but it doesn't seem to be working:
```
name: CI

on:
  workflow_dispatch:
  pull_request:
  push:
    branches:
      - main
    tags:
      - "v*.*.*"

jobs:
  industrial_ci:
    strategy:
      matrix:
        ROS_DISTRO: [iron]
    runs-on: ubuntu-latest
    env:
      CCACHE_DIR: "${{ github.workspace }}/.ccache" # directory for ccache (and how we enable ccache in industrial_ci)
      BASEDIR: "${{ github.workspace }}/services/autonomy"
      CC: "clang-20"
      CXX: "clang++-20"
      CLANG_FORMAT_VERSION: 14
    steps:
      - name: Checkout repository
        uses: actions/checkout@v4
      - name: Setup up ccache
        uses: actions/cache@v4 # fetch/store the directory used by ccache before/after the ci run
        with:
          path: ${{ env.CCACHE_DIR }}
          # This configuration will always create a new ccache cache starting off from the previous one (if any).
          # In this simple version it will be shared between all builds of the same ROS_REPO and ROS_REPO
          # and might need some fine-tuning to match the use case
          key: ccache-${{ matrix.ROS_DISTRO }}-${{github.run_id}}
          restore-keys: |
            ccache-${{ matrix.ROS_DISTRO }}-
      - name: Run industrial_ci
        uses: ros-industrial/industrial_ci@master
        env:
          ROS_DISTRO: ${{ matrix.ROS_DISTRO }}
          BASEDIR: ${{ env.BASEDIR }}
          BUILDER: colcon
          CCACHE_DIR: ${{ env.CCACHE_DIR }}
```
With gitlab I think the syntax is pretty clear, but I am unsure of the gh action.