ros-industrial/industrial_ci #882: Bitbucket legacy pipelines deprecated --> "authorization denied"

AndyZe, 2024-09-18:
On Sep 17 I started experiencing this issue on my bitbucket pipelines:

```
time="2024-09-18T19:00:50.149245108Z" level=error msg="AuthZRequest for HEAD /_ping returned error: authorization denied by plugin pipelines: "
```

I guess it's probably not a coincidence that bitbucket deprecated legacy pipelines on Sep 17.

There are some solutions described [here](https://community.atlassian.com/t5/Bitbucket-questions/Docker-build-failing-for-buildkit-with-error-authorization/qaq-p/2377667#M94515)

Basically it sounds like this is deprecated:  `DOCKER_BUILDKIT=0`

Maybe the new way to do it is:  `export PATH=/usr/bin:$PATH`

Here's my `bitbucket-pipelines.yml` with the fix I attempted. It didn't work.

```
image: docker:git

pipelines:
  default:
     - step:
         services:
           - docker
         script:
           # Attempted fix
           # See https://community.atlassian.com/t5/Bitbucket-questions/Docker-build-failing-for-buildkit-with-error-authorization/qaq-p/2377667#M94515
           - export PATH=/usr/bin:$PATH
           - apk add --update bash coreutils tar
           - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
           - .industrial_ci/bitbucket.sh ROS_DISTRO=humble

definitions:
  services:
    docker:
      memory: 2048
```

mathias-luedtke, 2024-09-19:
Thanks for reporting!

 > time="2024-09-18T19:00:50.149245108Z" level=error msg="AuthZRequest for HEAD /_ping returned error: authorization denied by plugin pipelines: "

Could you give me a little bit more context, please?
This looks like the "docker pull" fails or gets blocked.
You could set `TRACE=true` as well, which should print all the internal industrial_ci commands, which should help to narrow the scope.

> Basically it sounds like this is deprecated: DOCKER_BUILDKIT=0

This has been deprecated (by docker) for a while now.
And it does not matter, because industrial_ci does not build any docker images. It just runs them.

> Maybe the new way to do it is: export PATH=/usr/bin:$PATH

I would not mess with $PATH unless there is a good reason to do that.
To debug that more: Could you add an `env` call to your script and see what is the default for PATH.
And does bitbucket set any DOCKER_* environment variables?

mathias-luedtke, 2024-09-19:
According to https://support.atlassian.com/bitbucket-cloud/docs/run-docker-commands-in-bitbucket-pipelines/ running docker containers should be a supported use case.
`image: docker:git`  is not listed in any of the examples, though.

And it looks like bitbucket restricts the locations that can be mounted via volumes.
Please try setting `DOCKER_CREDENTIALS=""`. This should disable the extra-mounts, but as well the option to clone other repos.

If Docker does work anymore, there is as well the `ISOLATION=shell `option:
https://github.com/ros-industrial/industrial_ci/blob/master/.gitlab-ci.yml#L39

AndyZe, 2024-09-19:
I added this stuff to bitbucket-pipelines.yml. Didn't seem to change anything, though-

```

AndyZe, 2024-09-19:
Adding `export ISOLATION=shell` really screws things up:

```
image: docker:git
pipelines:
  default:
     - step:
         size: 2x # More memory
         services:
           - docker
         script:
           - export TRACE=true
           - env
           - export DOCKER_CREDENTIALS=""
           - export ISOLATION=shell
           - git clone git@bitbucket.org:apptronik/elevate_robotics.git
           - apk add --update bash coreutils tar
           - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
           - .industrial_ci/bitbucket.sh ROS_DISTRO=humble
definitions:
  services:
    docker:
      memory: 4096
```
-->

```
TRACE:util.sh:533 ici_guard source /opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src/tests/source_tests.sh
TRACE:util.sh:163 ici_hook before_init
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
init
/opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src/util.sh: line 466: debconf-set-selections: command not found
$ sudo sed -i /^# deb.*multiverse/ s/^# // /etc/apt/sources.list
sed: /etc/apt/sources.list: No such file or directory
'sudo sed -i /^# deb.*multiverse/ s/^# // /etc/apt/sources.list' returned with 1
TRACE:util.sh:316 ici_exit 1
   AT:util.sh:442 ici_error
   AT:util.sh:459 _ici_guard
   AT:workspace.sh:116 ici_cmd
   AT:util.sh:214 ici_init_apt
   AT:run.sh:61 ici_step
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
'init' returned with code '1' after 0 min 0 sec
TRACE:util.sh:208 ici_hook after_init
TRACE:ci_main.sh:75 ici_exit 
```

AndyZe, 2024-09-19:
This may all be a red herring. I have other bitbucket pipelines that are still succeeding, same `bitbucket-pipelines.yml`. Very confused...

mathias-luedtke, 2024-09-20:
According to your logs, industrial_ci is running and even managed to start the docker container and install things.
What happens, if you remove "image: docker:git"?

AndyZe, 2024-09-21:
Same symptoms unfortunately. The build tab is stuck at...

```
install_target_dependencies
...
Setting up ros-humble-moveit-resources-panda-moveit-config (2.0.7-1jammy.20240830.212942) ...
Setting up adwaita-icon-theme (41.0-1ubuntu1) ...
Setting up humanity-icon-theme (0.6.16) ...
Setting up ubuntu-mono (20.10-0ubuntu2) ...
Setting up libgtk-3-0:amd64 (3.24.33-1ubuntu2.2) ...
Setting up libgtk-3-bin (3.24.33-1ubuntu2.2) ...
Setting up qt5-gtk-platformtheme:amd64 (5.15.3+dfsg-2ubuntu0.2) ...
```

The docker tab is stuck at... (don't ask me why there's still a docker tab)
```
time="2024-09-21T16:47:04Z" level=info msg="Pipelines plugin request authorization." allowed=true method=POST plugin=pipelines uri=/v1.41/containers/2ca37a1066c1860ebd3ca85ec48b165e2943bfb1f3db83ee87ae710ec7a7dd84/start
time="2024-09-21T16:47:04.711954506Z" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
time="2024-09-21T16:47:04.712371367Z" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
time="2024-09-21T16:47:04.712393985Z" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
time="2024-09-21T16:47:04.712543646Z" level=info msg="loading plugin \"io.containerd.ttrpc.v1.pause\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
```

[image]

AndyZe, 2024-09-21:
Maybe relevant:

> I guess there is some sort of rate limiting mechanism in Bitbucket pipelines causing only the first few pulls to succeed from ecr. I solved the issue by starting SAM with --skip-pull-image option.

https://stackoverflow.com/q/78424795

AndyZe, 2024-09-21:
OK, if I remove some deps, I can at least get to the point where the tests are run. They don't pass because `controller_manager` package is missing but that verifies that it's a bitbucket API call limitation.

mathias-luedtke, 2024-09-22:
> OK, if I remove some deps, I can at least get to the point where the tests are run. They don't pass because controller_manager package is missing but that verifies that it's a bitbucket API call limitation.

For me it looks like 2h time limit for the pipeline.
(https://community.atlassian.com/t5/Bitbucket-questions/Increasing-pipeline-runner-time-limits/qaq-p/2013093)
industial_ci should only pull the image once and then start it.
How long does it take locally? (https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#run-industrial-ci-on-local-host)

> Is the docker image that industrial_ci uses based on ros-desktop-full? Maybe that would help?

No, industrial_ci uses` ubuntu:$codename` by default.
You can override it with ´DOCKER_IMAGE'.

AndyZe, 2024-09-22:
It only takes about 30s to build and another 30s to run tests locally. It's not a huge pkg at all.

OK, this got me to the point where CI quickly runs tests. I'm happy with that! Closing the issue, thanks for the hints.

```
         script:
           - export DOCKER_IMAGE="ros:humble"
```

closed 2024-09-23 by AndyZe