ros-industrial/industrial_ci #883: Bitbucket: setting of `UPSTREAM_WORKSPACE` causes the CI job to hang

AndyZe, 2024-09-24:
In `bitbucket-pipelines.yaml`, adding this line causes my pipeline to hang:

```
...
         script:
           - export TRACE=true
           - export DOCKER_IMAGE="ros:humble"
           # This line causes the pipeline to hang
           #- export UPSTREAM_WORKSPACE="bitbucket:my_org/aladdin_resources#main bitbucket:my_org/isaac_ros_cumotion#main github:PickNikRobotics/bio_ik#ros2"
           - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
           - .industrial_ci/bitbucket.sh ROS_DISTRO=humble
...
```

Specifically, the logs says it hangs here:

```
d92d07b72e74: Pull complete
bd82b3fc9b82: Pull complete
Digest: sha256:482ae18aa5d4813dd5c59aee9e4cd830eac94c60587f494e9ff343e6aaf3aba3
Status: Downloaded newer image for ros:humble
docker.io/library/ros:humble
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
'pull_docker_image' returned with code '0' after 0 min 14 sec
TRACE:util.sh:208 ici_hook after_pull_docker_image
$ docker create --init --env-file /opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src/isolation/docker.env --rm -v /opt/atlassian/pipelines/agent/build:/opt/atlassian/pipelines/agent/build:ro -e TARGET_REPO_PATH=/opt/atlassian/pipelines/agent/build -v /opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src:/opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src:ro -e ICI_SRC_PATH=/opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src -t --entrypoint  -w /opt/atlassian/pipelines/agent/build ros:humble /bin/bash /opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src/run.sh source_tests run_source_tests
TRACE:isolation/docker.sh:153 ici_guard docker inspect --format={{.Config.Image}} 8ebe7a8f38f06ba9e81cfdf2c7325c57152ec3277201690ab1602ac6f2d4272e
Copy credentials: /root/.ssh
$ docker start -a 8ebe7a8f38f06ba9e81cfdf2c7325c57152ec3277201690ab1602ac6f2d4272e
^ hangs here until the job times out
```

AndyZe, 2024-09-24:
I guess you'll tell me to set `- export VERBOSE_OUTPUT=true` so we can see what's going on...

AndyZe, 2024-09-24:
My guess is, it's a permissions issue with cloning the other repo in our org.

AndyZe, 2024-09-24:
OK, I've verified that it is a permissions issue with cloning the other repos in our org. Now need to find a way to circumvent this...

AndyZe, 2024-09-25:
I think what I need to do is place a private ssh key in the docker container somehow.

gavanderhoorn, 2024-09-25:
Another option could be to use a deploy token or other kind of token, and then use Git's `insteadOf` functionality ([docs](https://git-scm.com/docs/git-config#Documentation/git-config.txt-urlltbasegtinsteadOf)). You can use that to essentially 'redirect' `git clone`s of private repositories to URLs including the (deploy) token.

In case you do that by generating a `.gitconfig` file, you'd write that secret to disk, so consider the implications. You can however also pass Git config elements on the command line, for single invocations, which would avoid that. That would however perhaps be difficult to use with a `vcs` based workflow.

gavanderhoorn, 2024-09-26:
> I haven't tried what gavanderhoorn suggested yet. I'm not that desperate yet.

it's not such an awful way to do this :) I've seen it used in production at pretty large orgs.

Is it the prettiest? No. But depending on the exact circumstances (ephemeral containers on self-hosted infrastructure fi) it could be an acceptable way to get this to work.

Proper SSH setup would indeed be better though.

AndyZe, 2024-09-26:
OK, i got it working by cloning my private deps into the CI build/ folder before running the `.industrial_ci/bitbucket.sh` script  :+1:

image: docker:git
pipelines:
  default:
     - step:
         size: 2x # More memory
         services:
           - docker
         script:
           - export DOCKER_IMAGE="ros:humble"
           - apk add --update bash coreutils tar
           # Move aladdin_monitors_1 files from build/ to build/aladdin_monitors_1 so we can clone deps in the same ws
           - mkdir ./aladdin_monitors_1
           - mv $(ls --ignore=aladdin_monitors_1) ./aladdin_monitors_1
           # Clone deps
           # After this, we should have several packages in this build folder
           - git clone -b main git@bitbucket.org:my_org/aladdin_resources.git
           # Other dependencies go in the upstream workspace
           - export UPSTREAM_WORKSPACE="github:PickNikRobotics/bio_ik#ros2"
           - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
           - .industrial_ci/bitbucket.sh ROS_DISTRO=humble

closed 2024-09-26 by AndyZe