ros-industrial/industrial_ci #884: Local dependencies

Nir-Az, 2024-09-25:
Hi guys,
Some workflow question here:
We build a ROS2 package that is dependent on another ROS2 package we have.
When we change API the industrial_ci jobs fails because it does not use the new commit API of the dependent package but the published one.

So until we will publish the dependent package the flow will always fail.
Any W/A to this issue?

Thanks,
Nir

mathias-luedtke, 2024-09-30:
You could configure your UPSTREAM/DOWNSTREAM_WORKSPACE (https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#workspace-management) Perhaps with a custom script, which configures the branches as needed.

But in general I would recommend to not break API, but deprecate the old one first.

Nir-Az, 2024-10-06:
> You could configure your UPSTREAM/DOWNSTREAM_WORKSPACE (https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#workspace-management) Perhaps with a custom script, which configures [...]

I agree, but that's not the case here.
It's a new API for the dependency (new feature) that the ROS node will now use..
The only smooth way IMO is to release the dependency with the new API only, wait for sync and then add the new API at the ROS node (which is a bad W/A)..

Nir-Az, 2025-08-14:
Any suggestions?