ros-industrial/industrial_ci #901: How to run additional steps in the same environment

firesurfer, 2025-01-24:
Hi I am currently using the industrial_ci successfully in our github ci pipeline. 
Now I would like to add additional steps that are run in the same environment. In this specific case I would like to run xacro on our urdf.xacro files in order to test if certain configuration can be processed successfully. 

What I tried so far:

1.  Add an additional step after:

```
      - uses: 'ros-industrial/industrial_ci@master'
        env:
          UPSTREAM_WORKSPACE: ${{ inputs.upstream_workspace }}
          ROS_DISTRO: ${{ inputs.ros_distro }}
          ROS_REPO: ${{ inputs.ros_repo }}
      - name: Xacro
        run: |
            xacro <path to xacro file>
```

2. Use the AFTER_SCRIPT to run the same command

In both cases it looks like the command is not run in the same environment as the one the industrial_ci pipeline runs in. (command xacro not found)

So the question is:
How can I run a custom command in the same environment after the build has finished?

mathias-luedtke, 2025-02-02:
> How can I run a custom command in the **same** environment after the build has finished?

There is no shared  environment! All commands create their custom environment as needed.

> Add an additional step after:

This only works, if you run it on the target Ubuntu version with `ISOLATION: shell`

> Use the AFTER_SCRIPT to run the same command

This should work!

But in both cases you will have to source setup.bash, for example by prefixing your command with `rosenv`, see
https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#customize-within-the-ci-process for details.

firesurfer, 2025-02-07:
Thanks a lot for your answer!

> There is no shared environment! All commands create their custom environment as needed.

I wasn't really aware of that after reading the documentation but perhaps I just missed it. Especially `AFTER_SCRIPT` reads like it is executed in the same container. 

> https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#customize-within-the-ci-process

This perfectly explains it.

closed 2025-02-07 by firesurfer