ros-industrial/industrial_ci #919: All distros fail on init stage

Nir-Az, 2025-07-02:
Hi all,

Got an issue where all ROS2 distros fail on init stage, we tried taking the latest master and it looks the same.

Any idea why?
https://github.com/IntelRealSense/realsense-ros/pull/3374/checks

Thanks,
Nir

mathias-luedtke, 2025-07-02:
Your builds seem to have connectivity issues:

> W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/focal/InRelease  Connection failed [IP: 185.125.190.82 80]

https://github.com/IntelRealSense/realsense-ros/actions/runs/16027701288/job/45219827563#step:3:56

fmessmer, 2025-07-02:
canonical was down completely today, i.e. archive.ubuntu.com and others
see https://status.canonical.com

Nir-Az, 2025-07-03:
> [@Nir-Az](https://github.com/Nir-Az): have things improved in the meantime?

Yeah!
Sorry for the false alarm, thought it's an old docker image issue.

Looks better now
Thanks all

closed 2025-07-03 by Nir-Az